flightaware/dump1090 #65: Can't make: "Modes" defined multiple times

JinEnMok, 2020-06-06:
I'm getting the following error when trying to `make` the source (using GCC 10.1.0 on Arch Linux):

```
/usr/bin/ld: interactive.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: mode_ac.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: mode_s.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: comm_b.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: net_io.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: crc.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: demod_2400.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: stats.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: icao_filter.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: track.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: util.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: convert.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: sdr_ifile.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: sdr.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: sdr_rtlsdr.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
/usr/bin/ld: sdr_bladerf.o:/home/USER/Desktop/dump1090/dump1090.h:400: multiple definition of `Modes'; dump1090.o:/home/USER/Desktop/dump1090/dump1090.h:400: first defined here
collect2: error: ld returned 1 exit status
make: *** [Makefile:48: dump1090] Error 1
```

The original dump1090 by antirez builds fine. What could be the problem?

mutability, 2020-06-06:
Interesting; the linkage does look wrong, but .. it hasn't changed since 2013.

JinEnMok, 2020-06-06:
Could it be a compiler issue?

mutability, 2020-06-06:
Ah, here we go, it's a change in gcc 10:

https://gcc.gnu.org/gcc-10/porting_to.html: "default to -fno-common"

JinEnMok, 2020-06-06:
Thanks, that fixed it! I was able to compile the dev branch. I believe I can close the issue now?

mutability, 2020-06-06:
Yep, thanks for the report!

commit 0793c64e (2020-06-06): Clean up linkage of struct Modes to actually make sense.

(how did this work before? But it's been unchanged since at least
2013..)

Maybe fixes #65
closed 2020-06-06 by mutability