flightaware/dump1090 #68: Does not compile with GCC10

neheb, 2020-06-29:
```
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: interactive.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: mode_ac.o (symbol from plugin): in function `modeACInit':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: mode_s.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: comm_b.o (symbol from plugin): in function `decodeCommB':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: net_io.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: crc.o (symbol from plugin): in function `modesChecksum':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: demod_2400.o (symbol from plugin): in function `demodulate2400':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: stats.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: icao_filter.o (symbol from plugin): in function `icaoFilterInit':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: track.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: util.o (symbol from plugin): in function `mstime':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: convert.o (symbol from plugin): in function `init_converter':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: sdr_ifile.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: sdr.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
/home/mangix/devstuff/openwrt/staging_dir/toolchain-mips_74kc+dsp2_gcc-10.1.0_musl/lib/gcc/mips-openwrt-linux-musl/10.1.0/../../../../mips-openwrt-linux-musl/bin/ld: sdr_rtlsdr.o (symbol from plugin): in function `Modes':
(.text+0x0): multiple definition of `Modes'; dump1090.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
```

GCC10 uses -fno-common by default. This causes issues with an uninitialized global struct.

mutability, 2020-06-30:
This is already fixed on the dev branch, try that branch.

mutability, 2020-06-30:
Duplicate of #65

closed 2020-06-30 by mutability