flightaware/dump1090 #69: Skyview Frontend center map and receiver location displayed wrong in the map

glowfishDE, 2020-07-01:
This is referring to: https://github.com/Thom-x/docker-fr24feed-piaware-dump1090/issues/25

It seems to be a bug in the piaware skyview frontend not displaying the given coordinates (config.json) correctly but with an offset of a few hundred meters.

mutability, 2020-07-01:
`receiver.json` (generated by dump1090 based on the command-line-provided location) overrides whatever is in `config.js`
dump1090 defaults to emitting rounded coordinates for privacy reasons.
You can change this rounding behaviour via the `--json-location-accuracy` command-line arg.

mutability, 2020-07-01:
please reopen with specific details if that doesn't solve your issue

closed 2020-07-01 by mutability