flightaware/dump1090 #70: Unable to compile with GCC 9.3.0

assapir, 2020-07-12:
```
/usr/bin/ld: sdr_rtlsdr.o: in function `rtlsdrCallback':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:286: undefined reference to `rtlsdr_cancel_async'
/usr/bin/ld: sdr_rtlsdr.o: in function `rtlsdrRun':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:365: undefined reference to `rtlsdr_read_async'
/usr/bin/ld: sdr_rtlsdr.o: in function `rtlsdrClose':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:376: undefined reference to `rtlsdr_close'
/usr/bin/ld: sdr_rtlsdr.o: in function `rtlsdrOpen':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:171: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: sdr_rtlsdr.o: in function `find_device_index':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:97: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:116: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:124: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:132: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: sdr_rtlsdr.o: in function `show_rtlsdr_devices':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:82: undefined reference to `rtlsdr_get_device_count'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:87: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: sdr_rtlsdr.o: in function `rtlsdrOpen':
/home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:188: undefined reference to `rtlsdr_get_device_usb_strings'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:193: undefined reference to `rtlsdr_get_device_name'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:197: undefined reference to `rtlsdr_open'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:215: undefined reference to `rtlsdr_get_tuner_gains'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:222: undefined reference to `rtlsdr_get_tuner_gains'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:236: undefined reference to `rtlsdr_set_tuner_gain'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:240: undefined reference to `rtlsdr_get_tuner_gain'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:210: undefined reference to `rtlsdr_set_tuner_gain_mode'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:249: undefined reference to `rtlsdr_set_freq_correction'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:250: undefined reference to `rtlsdr_set_center_freq'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:251: undefined reference to `rtlsdr_set_sample_rate'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:253: undefined reference to `rtlsdr_reset_buffer'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:206: undefined reference to `rtlsdr_set_direct_sampling'
/usr/bin/ld: /home/alarm/.cache/yay/dump1090-fa-git/src/dump1090/sdr_rtlsdr.c:246: undefined reference to `rtlsdr_set_agc_mode'
```

GCC:
```
gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/armv7l-unknown-linux-gnueabihf/9.3.0/lto-wrapper
Target: armv7l-unknown-linux-gnueabihf
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://github.com/archlinuxarm/PKGBUILDs/issues --enable-languages=c,c++,fortran,go,lto,objc,obj-c++,d --enable-shared --enable-threads=posix --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --disable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --host=armv7l-unknown-linux-gnueabihf --build=armv7l-unknown-linux-gnueabihf --with-arch=armv7-a --with-float=hard --with-fpu=vfpv3-d16 gdc_include_dir=/usr/include/dlang/gdc
Thread model: posix
gcc version 9.3.0 (GCC)
```

Also same on dev branch

mutability, 2020-07-12:
You're missing suitable linker args to find librtlsdr. Usually these are picked up by pkg-config. See the Makefile.

assapir, 2020-07-13:
Thanks for the hint. Was fixed by using non-git `rtl-sdr` version

closed 2020-07-13 by assapir