flightaware/dump1090 #83: New flags not stackable?

shbisson, 2020-10-01:
So I have been experimenting with the new dump 1090 flags, and I don't seem to be able to stack them. I would like to be able to use the URL http://192.0.0.1/dump1090-fa/?banner=hide?aircraftTrails=show in the configuration for my twitterbot, but it doesn’t  work (and also seems to crash chromedriver in selenium). Anyone able to get this to work?

shbisson, 2020-10-01:
Thanks for the fix!

shbisson, 2020-10-01:
Ah, a comment on the FA forums notes that the documentation has an error, and stacking is & not ? which I should have noticed having been doing this web thing since the early 90s!

mutability, 2020-10-02:
Yep, it should just be the usual URL query parameter syntax with `&` separators

closed 2020-10-02 by mutability