flightaware/dump1090 #84: v4.0 upintheair.json not loaded

Lat31320, 2020-10-01:
Hi,

The /usr/share/dump1090-fa/html/upintheair.json file is no longer loaded with dump1090-fa 4.0 version despite the script.js (line 933) is checking for its presence.

File ACLs are ok
My "upintheair.json" file is the same for months

Site position is defined (and shown)

Regards.

mutability, 2020-10-02:
Can you try the fix in #82 and see if that helps?

Thom-x, 2020-10-02:
You type this into the console:
```javascript
var request = $.ajax({ url: 'upintheair.json',
                               timeout: 5000,
                               cache: true,
                               dataType: 'json' });
        request.done(function(data) {
                var ringStyle = new ol.style.Style({
                        fill: null,
                        stroke: new ol.style.Stroke({
                                color: '#000000',
                                width: 1
                        })
                });

                for (var i = 0; i < data.rings.length; ++i) {
                        var geom = new ol.geom.LineString([]);
                        var points = data.rings[i].points;
                        if (points.length > 0) {
                                for (var j = 0; j < points.length; ++j) {
                                        geom.appendCoordinate([ points[j][1], points[j][0] ]);
                                }
                                geom.appendCoordinate([ points[0][1], points[0][0] ]);
                                geom.transform('EPSG:4326', 'EPSG:3857');

                                var feature = new ol.Feature(geom);
                                feature.setStyle(ringStyle);
                                StaticFeatures.push(feature);
                        }
                }
        });
```

Lat31320, 2020-10-02:
> You can type this directly into the console to load properly upintheair:
> 
> ```js
> var request = $.ajax({ url: 'upintheair.json',
>                                timeout: 5000,
>                                cache: true,
>                                dataType: 'json' });
>         request.done(function(data) {
>                 var ringStyle = new ol.style.Style({
>                         fill: null,
>                         stroke: new ol.style.Stroke({
>                                 color: '#000000',
>                                 width: 1
>                         })
>                 });
> 
>                 for (var i = 0; i < data.rings.length; ++i) {
>                         var geom = new ol.geom.LineString([]);
>                         var points = data.rings[i].points;
>                         if (points.length > 0) {
>                                 for (var j = 0; j < points.length; ++j) {
>                                         geom.appendCoordinate([ points[j][1], points[j][0] ]);
>                                 }
>                                 geom.appendCoordinate([ points[0][1], points[0][0] ]);
>                                 geom.transform('EPSG:4326', 'EPSG:3857');
> 
>                                 var feature = new ol.Feature(geom);
>                                 feature.setStyle(ringStyle);
>                                 StaticFeatures.push(feature);
>                         }
>                 }
>         });
> ```

Indeed ! It's ok as instant temporary solution. Thanks.
I'm going to try the fix in #82 now

Lat31320, 2020-10-02:
> Can you try the fix in #82 and see if that helps?

The LineString fix do the trick ! Thanks a lot !

closed 2020-10-02 by eric1tran

Thom-x, 2021-03-15:
Bug reintroduced on v5.0

mutability, 2021-03-15:
@eric1tran

eric1tran, 2021-03-15:
@Thom-x  Can you try moving your upintheair.json into /usr/share/skyaware/html and let me know if that fixes it? We've created a separate map to merge 1090/978 together. I'll add it to the release notes for users using this feature.

Thom-x, 2021-03-15:
It's working for the new merged map.

eric1tran, 2021-03-15:
Thanks, I've added it to the release notes.