flightaware/dump1090 #85: UI showing _ infront of identities

diezel99, 2020-10-02:
[image]

mutability, 2020-10-03:
These are aircraft which are not transmitting a callsign; the registration with a leading `_` is shown instead.

diezel99, 2020-10-03:
For the life of me, I didn't see that stated anywhere.

closed 2020-10-03 by diezel99

mutability, 2020-10-03:
Yeah I don't think it is documented (and could be clearer)

mutability, 2020-10-04:
@eric1tran can we do something clearer here e.g. use parentheses or make it italic or something like that?

ericflightaware, 2020-10-05:
This is the same behavior as tar1090 and parallels the leading ~ formatting model for anonymous codes. A little in-line help/docs would go a long way here and could be used in a lot places.

eric1tran, 2020-10-05:
I'd like to find a good place for SkyAware usage documentation. I started this as a placeholder https://github.com/flightaware/dump1090/tree/master/public_html but www.flightaware.com or integrating it into SkyAware itself would be more ideal. 

We can do it for the next release as we add more features/changes.

mutability, 2020-10-05:
> parallels the leading ~ formatting model for anonymous codes

The leading `~` for non-ICAO addresses is an internal representation thing; it gets trimmed off by skyview and italics are used instead (at least in the table rows - the same really needs to be done in the aircraft detail pane too). We should do something similar for replacing ident with reg, a leading `_` is unclear because it looks like data from the aircraft itself and not a difference in data type.