flightaware/dump1090 #86: Error opening terminal in daemon mode

acasadoalonso, 2020-10-06:
When running with a daemon/service even with the **--quite** mode I get this error message:

dump1090.sh[12056]: Error opening terminal: unknown.

Full message from:                 **sudo service dump1090 status**

● dump1090.service - LSB: dump1090 initscript
   Loaded: loaded (/etc/init.d/dump1090.sh; generated)
   Active: active (exited) since Tue 2020-10-06 12:25:02 CEST; 37s ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 2200)
   Memory: 0B
   CGroup: /system.slice/dump1090.service

Oct 06 12:25:02 SoftRF systemd[1]: Starting LSB: dump1090 initscript...
Oct 06 12:25:02 SoftRF dump1090.sh[12056]: dump1090 started
Oct 06 12:25:02 SoftRF dump1090.sh[12056]: Tue Oct  6 12:25:02 2020 CEST  dump1090-fa OGN starting up.
Oct 06 12:25:02 SoftRF systemd[1]: Started LSB: dump1090 initscript.
Oct 06 12:25:02 SoftRF dump1090.sh[12056]: rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001000)
Oct 06 12:25:03 SoftRF dump1090.sh[12056]: Found Rafael Micro R820T tuner
Oct 06 12:25:03 SoftRF dump1090.sh[12056]: rtlsdr: enabling tuner AGC
Oct 06 12:25:03 SoftRF dump1090.sh[12056]: **Error opening terminal: unknown.**

mutability, 2020-10-06:
This appears to be a third-party repackaging of dump1090-fa; it's identifying itself as version "OGN", and we don't package an init.d script. You'd need to contact whoever built your package for support.

acasadoalonso, 2020-10-06:
Yes I built the program just on the invocation to the make file I called OGN instead of unknown ... it is just a comment 
With the unknown had the same behavior !!! 

AC/.

mutability, 2020-10-06:
Please explain exactly how you built and installed dump1090-fa, then.

acasadoalonso, 2020-10-06:
Very simple
Clone and

make DUMP_VERSION=OGN BLADERF=NO HACKRF=NO LIMESDR=NO

BTW: with the mutability version I had no problems whatsoever
The problem could be with the new rtlsdr libraries required to install 

AC/.

mutability, 2020-10-07:
Please explain exactly how you installed dump1090-fa.

(In particular there is no dump1090.sh supplied by dump1090-fa, so you need to explain what that is)

acasadoalonso, 2020-10-07:
Oliver,

Long time ago I wanted to use *dump1090 as a service*, I was using the
mutability version of dump1090 and I found the following shell script to
run it dump1090 as a service ... installed on the /etc/init.d/ directory.

When I installed the FlightAware version, I changed to point to the new
directory, however I found the error reported on this issue ... I wonder
now how you guys run *dump1090 as a service* ???

Thanks in advance for your help.
AC/.

--------------------------------------
#!/bin/bash
### BEGIN INIT INFO
#
# Provides: dump1090
# Required-Start: $remote_fs
# Required-Stop: $remote_fs
# Default-Start: 2 3 4 5
# Default-Stop: 0 1 6
# Short-Description: dump1090 initscript

#
### END INIT INFO
## Fill in name of program here.
PROG="dump1090"
PROG_PATH="/home/pi/src/dump1090.fa"
PROG_ARGS=" --interactive --net --quiet --no-fix --net-ro-size 500
--net-ro-rate 5 --net-heartbeat 60 --gain -10 --write-json /nfs/tmp "
PIDFILE="/var/run/dump1090.pid"

start() {
      if [ -e $PIDFILE ]; then
          ## Program is running, exit with error.
          echo "Error! $PROG is currently running!" 1>&2
          exit 1
      else
          ## Change from /dev/null to something like /var/log/$PROG if you
want to save output.
          cd $PROG_PATH
          ./$PROG $PROG_ARGS 2>&1 >/dev/null &
          echo "$PROG started"
          touch $PIDFILE
      fi
}

stop() {
      if [ -e $PIDFILE ]; then
          ## Program is running, so stop it
         echo "$PROG is running"
         killall $PROG
         rm -f $PIDFILE
         echo "$PROG stopped"
      else
          ## Program is not running, exit with error.
          echo "Error! $PROG not started!" 1>&2
          exit 1
      fi
}

## Check to see if we are running as root first.
## Found at http://www.cyberciti.biz/tips/shell-root-user-check-script.html
if [ "$(id -u)" != "0" ]; then
      echo "This script must be run as root" 1>&2
      exit 1
fi

case "$1" in
      start)
          start
          exit 0
      ;;
      stop)
          stop
          exit 0
      ;;
      reload|restart|force-reload)
          stop
          start
          exit 0
      ;;
      **)
          echo "Usage: $0 {start|stop|reload}" 1>&2
          exit 1
      ;;
esac
exit 0

mutability, 2020-10-07:
Your script tries to start dump1090 with `--interactive`

This makes no sense if you are trying to run it as a daemon and this option is the reason why dump1090 is trying (and failing) to open a terminal.

If you are on a Debian-like system, the recommended way of installing is to build and install as a Debian package - see the README. If you're on some other non-Debian system that's still using systemd, look at the systemd service file in `debian/dump1090-fa.service` as a starting point.

closed 2020-10-07 by mutability

acasadoalonso, 2020-10-07:
Apologies ... I did not realized that after building the Debian package, I need to do 
sudo dpkg -i dump1090-fa_4.0_armhf.deb 
Now everything works and thanks for your help !!!
 
executing:
 $ sudo service dump1090-fa status
● dump1090-fa.service - dump1090 ADS-B receiver (FlightAware customization)
   Loaded: loaded (/lib/systemd/system/dump1090-fa.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2020-10-07 13:25:50 CEST; 3min 31s ago
     Docs: https://flightaware.com/adsb/piaware/
 Main PID: 5523 (dump1090-fa)
    Tasks: 3 (limit: 2065)
   CGroup: /system.slice/dump1090-fa.service
           └─5523 /usr/bin/dump1090-fa --device-index 0 --gain -10 --ppm 0 --max-range 360 --fix --net --net-heartbeat 60 --net-ro-size 1300 --net-ro-interval 0.2 --net-ri-port 0 --net-

Oct 07 13:25:50  systemd[1]: Started dump1090 ADS-B receiver (FlightAware customization).
Oct 07 13:25:50  dump1090-fa[5523]: Wed Oct  7 13:25:50 2020 CEST  dump1090-fa 4.0 starting up.
Oct 07 13:25:50  dump1090-fa[5523]: rtlsdr: using device #0: Generic RTL2832U (Realtek, RTL2832U, SN 00001000)
Oct 07 13:25:51  dump1090-fa[5523]: Found Rafael Micro R820T tuner
Oct 07 13:25:51  dump1090-fa[5523]: rtlsdr: enabling tuner AGC
Oct 07 13:25:51  dump1090-fa[5523]: Allocating 4 zero-copy buffers

mutability, 2020-10-07:
I'll add a note to the README to make it clearer that you have to install the package after building it.