flightaware/dump1090 #91: how to let all flight listed and drawed in map

spid974, 2020-12-18:
Hello Dears , need help about 

Version 4 dump1090-fa (debian 10-64bit)
Question : how to let all fligth of 24h drawed in map and in list

Do I need to change values in /etc/default/dump1090-fa config files ? 

Thanks and best wishes

jankowskir, 2020-12-31:
You probably need something like Virtual Radar Server [https://www.virtualradarserver.co.uk/](url)

spid974, 2020-12-31:
Hello Dear, Thanks a lot, I will try collect data with VRS and let you know ! thanks again HNY

spid974, 2021-01-02:
I change value in Tools > Options ... > General 
But Remove from display & remove from tracking values can't be set more than 18000
I there a way to have aicrafts still tracked and drawed for 48h or more ?

mutability, 2021-01-04:
Skyaware isn't really aimed at long-term storage/display (storage is ephemeral and mostly on the browser side), try something like VRS as suggested, or tar1090; both can take data from your existing setup.

closed 2021-01-04 by mutability