flightaware/dump1090 #96: Not all data_validity structs are being expired

wiedehopf, 2021-01-14:
https://github.com/flightaware/dump1090/blob/b3fcf8247346b77e750950ad5cc40635b8cac35f/track.c#L1366

Might be on purpose?
In that case i can't see the purpose.

mutability, 2021-01-15:
Which are missing?

wiedehopf, 2021-01-15:
nac_v_valid
emergency_valid

Pretty sure there are more ... i didn't check specifically check your code, in my code just put them in ordered blocks so i could easily cross-check.

I suppose with your trackdataValid function it doesn't really matter and you could theoretically just get rid of the EXPIRE stuff.
(well trackDataAge would behave a bit different if you don't expire stuff)
```
283 static inline int trackDataValid(const data_validity *v)                                                                            
284 {
285     return (v->source != SOURCE_INVALID && messageNow() < v->expires);
286 }
```

In contrast it does matter in my fork as i changed up the trackDataValid function.
Just thought i'd let you know.

mutability, 2021-01-15:
Looks like it was just those two. I noticed that `emergency_valid` was not getting set up during aircraft track creation either. I think they'll all harmless since as you say the expiry time is already looked at, but no harm fixing it. Thanks for the heads-up.

commit c7675b3c (2021-01-15): Add missed expiry code for emergency / nac_v fields; initialize emergency_valid.

(noticed by @wiedehopf)

Fixes #96
closed 2021-01-16 by wiedehopf