flightaware/dump1090 #103: The dsp/benchmarks won't compile with clang

gtjoseph, 2021-02-01:
Lots of alignment complaints.  Not a big deal for now.  I was getting better performance in the converter and hirate demod with clang but starch may make that a moot point.

```
In file included from dsp/generated/benchmark.c:1259:
dsp/generated/../benchmark/magnitude_sc16_benchmark.c:11:16: error: '_Alignof' applied to an expression is a GNU extension [-Werror,-Wgnu-alignof-expression]
    if (!(in = STARCH_BENCHMARK_ALLOC(len, *in)) || !(out_mag = STARCH_BENCHMARK_ALLOC(len, *out_mag))) {
               ^
dsp/generated/benchmark.c:1256:81: note: expanded from macro 'STARCH_BENCHMARK_ALLOC'
#define STARCH_BENCHMARK_ALLOC(_count, _type) starch_benchmark_aligned_alloc(1, alignof(_type), (_count) * sizeof(_type))
                                                                                ^
/usr/lib64/clang/11.0.0/include/stdalign.h:15:17: note: expanded from macro 'alignof'
#define alignof _Alignof
                ^
```

mutability, 2021-02-02:
`decltype` would be nice here but it's C++ only. `typeof` is another gcc extension, I wonder if clang is happier about that. Otherwise we'll need to explicitly provide the type name.

mutability, 2021-02-02:
clang doesn't like `typeof`, so I'll just switch to explicit typenames

commit 2c7b3313 (2021-02-02): Pass literal type names to STARCH_BENCHMARK_ALLOC (fixes #103)
closed 2021-02-02 by mutability