flightaware/dump1090 #106: stats.json is truncated after about 60 seconds and is rarely written to again

gtjoseph, 2021-02-07:
With the new stats output, stats.json gets written to at startup but at about 60 seconds gets truncated.   Very rarely, the stats will appear again for one cycle, then the file is truncated again.  Tried with --json-stats-every values at 10, 60, and 70 seconds.

* With --json-stats-every set to 10, the initial stats will be written with all zeros, then they'll get updated 6 times, then the file will be truncated.
* With --json-stats-every set to 70, the initial stats will be written with all zeros, then the file will be truncated after 60 seconds with no valid data ever written.

mutability, 2021-02-07:
Truncation means the internal buffer is filling up, try increasing the buffer size in generateStatsJson

mutability, 2021-02-07:
testing note: need to run with `--net` to provoke this; without the remote-messages stats the stats output stays under 4k

mutability, 2021-02-07:
Should be fixed on the dev branch now.

I couldn't find anything wrong with `--net-only --json-stats-every 10`, it correctly updated the json file every 10 seconds for me.
`

gtjoseph, 2021-02-07:
Part of the problem was I was looking at last1min instead of latest because latest wasn't being populated before this. :)
Anyway, now getting good stats on time.  Thanks!

gtjoseph, 2021-02-07:
Sorry, Just as I saved that comment I noticed that with a --json-stats-every of 10 seconds, every 60 seconds, the latest stats are all zeros.  Maybe the output of the json stats has to happen _before_ the 60 second reset?

mutability, 2021-02-08:
That is working correctly. The "latest" stats cover the period from the end of the last 1-minute period to the current time. If the last 1-minute period just ended, the latest stats will be zero because there's no additional data after the last 1-minute period to show. Look at the start/end timestamps of the latest & 1-minute periods to verify.

gtjoseph, 2021-02-08:
OK, that's _very_ confusing.  I guess I was expecting that if I used `--json-stats-every 10` that the stats were for the last 10 seconds, not just a snapshot every 10 seconds of an arbitrary 60 second window.  So now how does this relate to last1min?

I guess it's OK for now though.  I haven't looked at the implementation because I've already got 6 branches going with the hirate demod and airspy stuff which I'm trying to get up to you in the next few days. :)

mutability, 2021-02-08:
The `stats.json` behaviour here has not changed - it's just that the `latest` numbers have been always-zero for a long time now, dating back to when the internal webserver was removed. When the internal webserver still existed and you could request `stats.json` at arbitrary times, you would have seen exactly the same behavior when requesting a copy at a time that didn't line up with a multiple of 60 seconds: `latest` held the partially accumulated stats for the currently-in-progress 1 minute windo (and so had a duration < 1 minute); `1min` held the last _completed_ 1 minute window.

`--json-stats-every` is just re-exposing that existing behaviour, since it lets you get a copy of `stats.json` that isn't aligned with the 1-minute window edge.

I don't see that exposing a resets-every-N-seconds set of stats is hugely useful, because it's not really any different to the current resets-every-60-seconds cycle and the client would additionally need to know the length of the cycle to do anything useful with it. What are you trying to do that requires a different reset cycle?

mutability, 2021-02-08:
see also https://github.com/flightaware/dump1090/blob/master/README-json.md#statsjson which documents this:

> * "last1min" covers a recent 1-minute period. This may be up to 1 minute out of date (i.e. "end" may be up to 1 minute old).
> [...]
> * "latest" covers the time between the end of the "last1min" period and the current time.

mutability, 2021-02-08:
re hi-rate demodulator, you may want to have a poke some early work I've pushed here: https://github.com/flightaware/dump1090/blob/pxsdr/demod_multi.inc

(it seems to work well so far; hooking it up to an airspy is on my list somewhere)

gtjoseph, 2021-02-08:
> What are you trying to do that requires a different reset cycle?

Compare 3 instances of dump1090-fa in real time.   One with an rtlsdr, one with airspy_adsb and one with the hirate demod and sdr_airspy. :)

[image]

I'm fine with the way it is.

> re hi-rate demodulator, you may want to have a poke some early work I've pushed here:

Thanks, I'll take a look.

commit 1b1f9de1 (2021-02-07): Increase default stats json buffer size to 8k. Retry with a larger buffer if we run out of space, like we do with aircraft.json

With the addition of the per-DF stats, the stats output started to exceed 4k in some cases (notably, you need --net turned on).

Fixes #106
closed 2021-02-08 by gtjoseph