flightaware/dump1090 #113: view1090 ground status for UAT targets

rcarsey1, 2021-02-18:
I have observed a case where view1090 reports that an aircraft is on "grnd" after it has departed.  The case seems reproducible if the aircraft has a UAT transponder.

My setup is running dump978-fa which is piped to uat2esnt.  This is then sent to ModeSMixer2 which combines my UAT data and ES1090 data and outputs it to a port (beast format).  View1090 reads the output feed from ModeSMixer2. 

The data looks correct when viewed in view1090 non-interactive mode.  I observe "Extended Squitter (Non-Transponder) Surface position (8)" frames.  The aircraft is displayed correctly in interactive mode on the ground.  Eventually I observe " Extended Squitter (Non-Transponder) Airborne position (barometric altitude) (18) (reliable)" frames (and velocity frames too).  The altitude in interactive mode remains "grnd".   uat2esnt is the party responsible for setting DF=18 (non-transponder).

In mode_s.c,  mm->airground is only set when mm->msgtype is 11 or 17  (not 18).  And later on, in  track.c, a->airground never gets updated and remains AG_GROUND.

I've attached a sample Beast capture with N317TA being the aircraft of interest.
[departure-N317TA.zip](https://github.com/flightaware/dump1090/files/5999659/departure-N317TA.zip)

mutability, 2021-02-18:
> In mode_s.c, mm->airground is only set when mm->msgtype is 11 or 17 (not 18).

I guess you're looking at https://github.com/flightaware/dump1090/blob/d8d5f2a1178d104e81ef84e2ff4af6ec15333b28/mode_s.c#L607

The CA field is only present in DF11/DF17. On DF18 those bits are used for the CF field instead.

airground can be set to AG_GROUND here based on a DF18 surface position: https://github.com/flightaware/dump1090/blob/d8d5f2a1178d104e81ef84e2ff4af6ec15333b28/mode_s.c#L936

There is deliberately no equivalent code to set `AG_AIRBORNE` from airborne positions. The diificulty is that "airborne" positions are actually not reliably airborne; they're also transmitted by aircraft that don't have a WoW sensor or otherwise can't detect air/ground (the spec is explicit about this). So airborne actually means "airborne or on the ground" in many cases and that is why AG_UNCERTAIN exists. Treating those "uncertain" cases as "definitely airborne" breaks processing of lots of real 1090-transmitting aircraft, so that's not really an option.

What I would expect to see happen in this case is:

 * surface positions stop getting sent. `a->airground == AG_GROUND`
 * subsequent messages have `AG_UNCERTAIN`. Tracked state stays with `a->airground == AG_GROUND`.
 * 15 seconds later, `airground_valid` passes its expiry time and the tracked airground state becomes stale
 * subsequent messages with `AG_UNCERTAIN` then update tracked state to `a->airground == AG_UNCERTAIN`
 * view1090 should stop showing "grnd" and start displaying altitudes at this point

Let me take a look at your capture and see if anything is misbehaving here.

mutability, 2021-02-18:
OK, so the cause is that DF18 airborne-position messages do not set `airground` at all (i.e. it is `AG_INVALID`, not `AG_UNCERTAIN`) so the tracked air/ground state is never updated. (With no updates, eventually that state will expire entirely, and you'll start seeing altitudes in interactive mode, but that takes 60 seconds not 15).

It should be safe to set `AG_UNCERTAIN` when processing an airborne position message with a valid altitude that doesn't have an existing `airground` value (i.e. DF18 or unusual DF17 CA value); that should fix the problem.

Also I'll update interactive mode to use `trackDataFresh` not `trackDataValid` when displaying the `grnd` override.

mutability, 2021-02-18:
In retrospect the interactive mode change isn't a great idea, since it will switch from grnd to altitude mode after 15 seconds if we entirely stop receiving data from the aircraft, so I'll skip that. Setting `AG_UNCERTAIN` should be sufficient.

mutability, 2021-02-18:
Your capture seems to behave correctly with that change:

` A3627C S     1200  N317TA    -100   48  045   40.329  -74.348 -50.0    97 21`

closed 2021-02-18 by mutability

rcarsey1, 2021-02-18:
I got (expectedly) the same results as you with the new code.  All of the local airport's school aircraft are UAT, so I'll have some more real-life data this weekend.  I've been hacking at the code a bit in order to record the daily number of  departures & arrivals (aircraft transitioning from ground-to-air and air-to-ground)  to a MySQL db.  Having flying aircraft still marked as `AG_GROUND` was troublesome.  Thanks for the assist!

mutability, 2021-02-18:
You might want to look at directly processing the UAT data from dump978-fa (the json message format is probably the simplest to handle) since uat2esnt is a fairly blunt instrument and detail is lost.