flightaware/dump1090 #114: Improve UI on mobile devices, vertical design

e2jk, 2021-02-27:
The UI works great on large screens like laptops, but is not ideal on smaller screens like smartphones.
For example this is how it shows on my 6" screen Android smartphone:

[image]

Main issues:
- Map is too narrow and long on the right side
- Text information in table format on the left is not readable without zooming
- Often zooming in causes issues when trying to zoom back out, in particular if the map is now full-screen-ish, since the pinching action then acts on the zoom level of the map instead of the entire page

On large screens, the screen real estate is most often horizontal, where it makes sense to have a design that lays out the map and the table content side-by-side.
But mobile devices are most often held in vertical orientation, so it would make sense to have the map and the table be laid out in vertical form, either with the table first or the map first.

Here a couple test mockups for a vertical design:

Vertical mockup, table first | Vertical mockup, map first
------------ | -------------
[image] | [image]

Any suggestions on which of the table or the map would make sense to be first?
I think I prefer the table first, but it might make for a more consistent layout to have the map first: when there is a large number of planes in view, having the table down would mean the table could continue below and allow the user to scroll down, with the map always consistently positioned on top?

Note: Ideally, the design would adapt itself to the orientation of the device, and switch back to the same side-by-side layout as on larger screens when the device is held horizontally.

eric1tran, 2021-03-15:
I like the vertical design. There's been discussion about making it a bit more mobile friendly for some upcoming projects we're working on so I'll pass the suggestion along.

gvanem, 2021-06-30:
I'll ask here instead of opening another issue.

I've built Dump1090-FA for Windows (with some patches and tweaks). It work great for `--interactive` mode! 
But I fail to see there is a Web-server built-in. An option somehow? Only SBS, Beast etc.
Does it mean I have to feed via RadarScape to server a Web-client?

mutability, 2021-06-30:
You should open a new issue for a new issue.

Dump1090 does not have an internal webserver, use --write-json and serve
the results through a standalone webserver.

raccettura, 2021-08-20:
> I like the vertical design. There's been discussion about making it a bit more mobile friendly for some upcoming projects we're working on so I'll pass the suggestion along.

Along with this, any chance on a query arg to show just the map or just the table? It would be helpful when embedding as an iframe into other interfaces like HomeAssistant.

sigwx, 2021-08-20:
I belive there is a
sidebar=hide
to hide the table, and
banner=hide
to hide the banner across the top. And you should be left with just the map

raccettura, 2021-08-20:
Just tested and that works.  Thanks so much!  Wish that little tidbit was better documented.

therealhalifax, 2023-01-11:
> Just tested and that works. Thanks so much! Wish that little tidbit was better documented.

How you added this to Home Assistant Addon? If I call the Webpage outside of Home Assistant with that arguements, it works also fine for me. But I need that inside the Addon, so that the agrs are always used when the Addon Frontend is called.

gregdev, 2024-02-22:
I've just submitted #235 which brings in improved display on smaller devices, notably vertical stacking of the map and info table. Currently I have the map sitting first and the table underneath but I do like the idea of this order being user-selectable. I'm also thinking I should make just the table rather than the whole window scroll, so that the map remains fixed in place - any thoughts on this?

PR #235 (2025-01-22): Mobile display improvements
I often like to quickly check my web interface on my phone, but the display on narrow screens isn't ideal. I've modified the markup and styles so that the map and info table are stacked vertically, plus a bunch of other tweaks to get things fitting in nicely.

Ultimately I'd love to completely redo the UI, and may get around to it one day, but this at least makes things usable on mobile.

Closes #114.