flightaware/dump1090 #115: Indicate the number of tracked planes at the beginning of the title

e2jk, 2021-02-27:
The page title contains the number of planes currently seen by the local installation, but those numbers are shown at the end of the title, meaning it is often not visible without explicitly hovering on the tab, in particular if you have a larger number of tabs open.

[image]

Wouldn't it be possible/better to have those numbers show upfront, ensuring they are always visible? This would allow a user to know how many planes are in sight of the radar without needing to open the tab, while doing stuff on other tabs it would suffice to glance at the tab title to see if planes are currently in sight or not.

A concern could be that a user might be confused on what that tab is, if it shows only numbers and no name.
I don't think this should be a major problem though: users that have a dump1090 page open likely don't have this "by chance", it's probably their own radar they keep an eye on, or at least it's someone interested in aviation tracking and would soon realize the "&lt;number&gt;/&lt;number&gt;" tab is their plane tracking tab...

eric1tran, 2021-02-28:
Which dump1090 version are you on? I made a change to 4.0 to put the number of aircraft at the beginning of the tab text.

e2jk, 2021-02-28:
Thanks @eric1tran for having made that change (da3557fcfe83ef683b75f976f38a77efdd31be74).
My current version is v3.8.1, as I'm running from the `latest` tagged image of the [thomx/fr24feed-piaware](https://hub.docker.com/layers/thomx/fr24feed-piaware/1.3.1/images/sha256-4e09b38fad0db1079e387eccddf4414fc5fc67eb21996ec482be23b02f438ca1?context=explore) Docker image.

I do see there is a [thomx/fr24feed-piaware:flightaware-v4.0](https://hub.docker.com/layers/thomx/fr24feed-piaware/flightaware-v4.0/images/sha256-031f66bb4cba42420b58a68edbee0ab20ed65e604d9f950b0ec1d583f7e330fe?context=explore) tag, but it's older than the latest and is only compiled for amd64, I'll compile it on my armv6 Rapsberry Pi to test this out.

Closing this issue as I believe it is fixed in the latest version, thanks.

closed 2021-02-28 by e2jk