flightaware/dump1090 #121: skyaware new WebUI issue

DenyDarko, 2021-03-17:
Hey there,
The button shown on the screenshot does nothing:

[image]

Is that expected?

eric1tran, 2021-03-17:
That button should only show up for SD card installs since it includes piaware-web and exposes status.json which contains a link to your stats page. Are you seeing it on yours? Do you have piaware-web installed by chance?

DenyDarko, 2021-03-17:
> That button should only show up for SD card installs since it includes piaware-web and exposes status.json which contains a link to your stats page. Are you seeing it on yours? Do you have piaware-web installed by chance?

Nope:

`root@raspberrypi:~ # apt list --installed pia*
Listing... Done
piaware/now 5.0 arm64 [installed,local]
root@raspberrypi:~ #
`

closed 2022-09-27 by eric1tran