flightaware/dump1090 #124: Compiling on Ubuntu 20.04 x86_64 Generates Errors

phipac, 2021-03-30:
I will do my best to not frustrate any of you. I am a relative newbie when it comes to coding. I am having difficulties getting the install script to work on Ubuntu 20.04 x86_64. I usually see folks do not post enough information, so I am going to attempt to give you everything you need.

uname -a
Linux name 5.8.0-48-generic #54~20.04.1-Ubuntu SMP Sat Mar 20 13:40:25 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

/dump1090$ sudo dpkg-buildpackage -b --no-sign
dpkg-buildpackage: info: source package dump1090-fa
dpkg-buildpackage: info: source version 5.0
dpkg-buildpackage: info: source distribution stable
dpkg-buildpackage: info: source changed by Eric Tran <eric.tran@flightaware.com>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source --before-build .
 debian/rules clean
dh clean --with=systemd
dh: warning: Compatibility levels before 10 are deprecated (level 9 in use)
   dh_auto_clean
dh_auto_clean: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        make -j1 clean
make[1]: Entering directory '/home/sfixphdi/dump1090'
rm -f *.o oneoff/*.o compat/clock_gettime/*.o compat/clock_nanosleep/*.o cpu_features/src/*.o dsp/generated/*.o d                                                                                                                            sp/helpers/*.o cpu_features/src/filesystem.o cpu_features/src/stack_line_reader.o cpu_features/src/string_view.o                                                                                                                             cpu_features/src/hwcaps.o cpu_features/src/cpuinfo_x86.o dump1090 view1090 faup1090 cprtests crctests oneoff/conv                                                                                                                            ert_benchmark oneoff/decode_comm_b oneoff/dsp_error_measurement oneoff/uc8_capture_stats starch-benchmark
make[1]: Leaving directory '/home/sfixphdi/dump1090'
   dh_clean
        rm -f debian/debhelper-build-stamp
        rm -rf debian/.debhelper/
        rm -f debian/dump1090-fa.debhelper.log
dh_clean: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        rm -f -- debian/dump1090-fa.substvars debian/files
        rm -fr -- debian/dump1090-fa/ debian/tmp/
        find .  \( \( \
                \( -path .\*/.git -o -path .\*/.svn -o -path .\*/.bzr -o -path .\*/.hg -o -path .\*/CVS -o -path                                                                                                                             .\*/.pc -o -path .\*/_darcs \) -prune -o -type f -a \
                \( -name '#*#' -o -name '.*~' -o -name '*~' -o -name DEADJOE \
                 -o -name '*.orig' -o -name '*.rej' -o -name '*.bak' \
                 -o -name '.*.orig' -o -name .*.rej -o -name '.SUMS' \
                 -o -name TAGS -o \( -path '*/.deps/*' -a -name '*.P' \) \
                \) -exec rm -f {} + \) -o \
                \( -type d -a -name autom4te.cache -prune -exec rm -rf {} + \) \)
 debian/rules build
dh build --with=systemd
dh: warning: Compatibility levels before 10 are deprecated (level 9 in use)
   dh_update_autotools_config
   dh_auto_configure
dh_auto_configure: warning: Compatibility levels before 10 are deprecated (level 9 in use)
   debian/rules override_dh_auto_build
make[1]: Entering directory '/home/sfixphdi/dump1090'
dh_auto_build -- RTLSDR=yes BLADERF=yes HACKRF=yes LIMESDR=yes DUMP1090_VERSION=5.0
dh_auto_build: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        make -j1 RTLSDR=yes BLADERF=yes HACKRF=yes LIMESDR=yes DUMP1090_VERSION=5.0
make[2]: Entering directory '/home/sfixphdi/dump1090'
Building with:
  Version string:  5.0
  DSP mix:         x86
  RTLSDR support:  yes
  BladeRF support: yes
  HackRF support:  yes
  LimeSDR support: yes
cc -Wdate-time -D_FORTIFY_SOURCE=2 -I. -DMODES_DUMP1090_VERSION=\"5.0\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\"                                                                                                                             -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_HACKRF -D                                                                                                                            ENABLE_LIMESDR -DSTARCH_MIX_X86 -g -O2 -fdebug-prefix-map=/home/sfixphdi/dump1090=. -fstack-protector-strong -Wfo                                                                                                                            rmat -Werror=format-security -std=c11 -O3 -g -Wall -Wmissing-declarations -Werror -W -D_DEFAULT_SOURCE -fno-commo                                                                                                                            n -I/usr/local/include/ -I/usr/include/ -I/usr/include/ -I/usr/include/libusb-1.0  -c dump1090.c -o dump1090.o
cc -Wdate-time -D_FORTIFY_SOURCE=2 -I. -DMODES_DUMP1090_VERSION=\"5.0\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\"                                                                                                                             -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_HACKRF -D                                                                                                                            ENABLE_LIMESDR -DSTARCH_MIX_X86 -g -O2 -fdebug-prefix-map=/home/sfixphdi/dump1090=. -fstack-protector-strong -Wfo                                                                                                                            rmat -Werror=format-security -std=c11 -O3 -g -Wall -Wmissing-declarations -Werror -W -D_DEFAULT_SOURCE -fno-commo                                                                                                                            n -I/usr/local/include/ -I/usr/include/ -I/usr/include/ -I/usr/include/libusb-1.0  -c anet.c -o anet.o
cc -Wdate-time -D_FORTIFY_SOURCE=2 -I. -DMODES_DUMP1090_VERSION=\"5.0\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\"                                                                                                                             -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_HACKRF -D                                                                                                                            ENABLE_LIMESDR -DSTARCH_MIX_X86 -g -O2 -fdebug-prefix-map=/home/sfixphdi/dump1090=. -fstack-protector-strong -Wfo                                                                                                                            rmat -Werror=format-security -std=c11 -O3 -g -Wall -Wmissing-declarations -Werror -W -D_DEFAULT_SOURCE -fno-commo                                                                                                                            n -I/usr/local/include/ -I/usr/include/ -I/usr/include/ -I/usr/include/libusb-1.0  -c interactive.c -o interactiv                                                                                                                            e.o
cc -Wdate-time -D_FORTIFY_SOURCE=2 -I. -DMODES_DUMP1090_VERSION=\"5.0\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\"                                                                                                                             -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_BLADERF -DENABLE_HACKRF -D                                                                                                                            ENABLE_LIMESDR -DSTARCH_MIX_X86 -g -O2 -fdebug-prefix-map=/home/sfixphdi/dump1090=. -fstack-protector-strong -Wfo                                                                                                                            rmat -Werror=format-security -std=c11 -O3 -g -Wall -Wmissing-declarations -Werror -W -D_DEFAULT_SOURCE -fno-commo                                                                                                                            n -I/usr/local/include/ -I/usr/include/ -I/usr/include/ -I/usr/include/libusb-1.0  -c mode_ac.c -o mode_ac.o
cc -Wdate-time -D_FORTIFY_SOURCE=2 -I. -DMODES_DUMP1090_VERSION=\"5.0\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\"                                                                                                                             -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_BL [... cut ...]
make[2]: Leaving directory '/home/sfixphdi/dump1090'
dh_auto_build: error: make -j1 RTLSDR=yes BLADERF=yes HACKRF=yes LIMESDR=yes DUMP1090_VERSION=5.0 returned exit c                                                                                                                            ode 2
make[1]: *** [debian/rules:48: override_dh_auto_build] Error 255
make[1]: Leaving directory '/home/sfixphdi/dump1090'
make: *** [debian/rules:57: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2

mutability, 2021-03-30:
It looks like you have a manually installed librtlsdr installed in
/usr/local/lib that is interfering with the package build. You'll need to
remove that for the package build to work.

phipac, 2021-04-01:
I had too many builds from previous projects that I forgot about. All is working well now. Thank you very much for your time!

-Phil

closed 2021-04-01 by phipac

Carpintonto, 2021-07-26:
Instead of removing librtlsdr installed in /usr/local/lib
we can append a directive in debian/rules:

```
override_dh_shlibdeps:
        dh_shlibdeps --dpkg-shlibdeps-params=--ignore-missing-inf
```

A tab must lead the second line, just format it the same as the two other override_dh directives

Many beginners including myself start using the the [RTL2832U Osmocom drivers](https://git.osmocom.org/rtl-sdr/), 
which depend on libusb-1.0-0-dev and that's where apt installs.