flightaware/dump1090 #128: Unable to build on Mac OSX BigSur - fix provided

apostigl, 2021-05-14:
I was unable to "make" on Mac OSX BigSur and XCode 12.5 unless changing the following files:

magnitude_sc16q11.c
magnitude_sc16.c
magnitude_power_uc8.c
magnitude_uc8.c

removing: `#include<endian.h>` and replacing it with

```
#if defined(__APPLE__)
#include <libkern/OSByteOrder.h>
#include <machine/endian.h>
#define htole16(x) OSSwapHostToLittleInt16(x)
#define htole32(x) OSSwapHostToLittleInt32(x)
#define le16toh(x) OSSwapLittleToHostInt16(x)
#define le32toh(x) OSSwapLittleToHostInt32(x)
#else
#include <endian.h>                     /* htole16, htole32, le32toh */
#endif
```

mutability, 2021-07-07:
Can  you check if b3172181d5f3ab2195dd7affd3e2882d31cf2403 on the dev branch fixes this for you?
(I don't have a OSX build environment right now)

mutability, 2021-08-04:
assuming fixed, please re-open if not.

commit b3172181 (2021-07-07): Use compat/compat.h for endian-swapping functions in DSP code.

Might fix #128
closed 2021-08-04 by mutability