flightaware/dump1090 #129: dump1090 not writting json files

carroarmato0, 2021-05-17:
I've had mixed results with dump1090 producing json files when using --write-json .
I'm not certain if its permissions related or if it's picky about relative or absolute paths, but adding an error message would make it easier to troubleshoot the issue.

carroarmato0, 2021-05-17:
I've figure out that it was permission issues. But having some debugging would have helped found out sooner :)

mutability, 2021-05-19:
IIRC the original reason for silently suppressing errors was that they were very noisy in the face of misconfiguration (either you make the first error encountered make dump1090 bail out entirely, or you spam lots of errors into the logs - neither is really ideal). But I agree that silently eating errors is not great; let's see what I can do to find a middle ground.

carroarmato0, 2021-05-19:
Shooting out errors on the first go encourages fixing the issue to get beautiful output again :)
There's also the use case of the filesystem getting full which you might not see immediately.

mutability, 2021-05-19:
If you fill up the filesystem is it preferable to keep logging that you filled up the filesystem, once a second, or to have dump1090 exit and restart? neither seems ideal..

carroarmato0, 2021-05-19:
If the filesystem is already full, re-attempting to log will fail anyway until things have been cleaned up or storage has been extended, in which case the program can continue to log, and no further action is necessary. 
It's the lesser evil :)

carroarmato0, 2021-05-19:
But I think permission denied is different, and the application should come to a halt until it's fixed because spamming the filesystem with "Write Permission error" is filling up the filesystem for nothing and the user does not get the expected result that things aren't logged.

mutability, 2021-08-04:
I took the middle ground and log-and-continue with rate limiting to no more than one message per minute:

```
$ ./dump1090 --write-json no-perms --net-only
Wed Aug  4 17:34:41 2021 +08  dump1090-fa unknown starting up.
Net-only mode, no SDR device or file open.
failed to create no-perms/receiver.json.Nr1pMT (for updating no-perms/receiver.json): Permission denied
failed to create no-perms/stats.json.nqseqU (for updating no-perms/stats.json): Permission denied (66 more error messages suppressed)
```

closed 2021-08-04 by mutability

carroarmato0, 2021-08-04:
Sounds good! Thanks.

commit ef703741 (2021-08-04): Report json write errors, with some rate limiting.

Fixes #129