flightaware/dump1090 #150: SoapyRemote support

jedi7, 2021-09-01:
Hi, 
it will be nice to have soapy remote support. So the RTL-SDR dongle can be shared with more software.

gvanem, 2021-09-13:
There is an old `sdrsoapy` branch in this project. Have you tried that?

jedi7, 2021-09-13:
As you mentioned, it is old branch. Maybe somebody knows why it is not merged.

mutability, 2021-09-14:
Just never got cleaned up to the point where it was ready, and I forget what the reason I was looking at it was..

The old branch is a reasonable starting point if someone wants to clean it up / test it / submit a PR

rcarsey1, 2021-11-27:
I'll second Soapy support.  I'm moving away from RTL dongles and onto SDRplay hardware which is a little more robust with their 14-bit ADC (while still being reasonably affordable).  My issue is that my antennas sit 30 feet from a runway and I must turn the gain down to almost 0 so I don't get overpowered.. which kills the receive distance to just a few miles.  The better dynamic range of 14-bit helps.

The end idea being that the RSPduo is one device that can do both UAT (dump978-fa already supports SoapySDR and seems to work OK) and dump1090-fa (which.. I'll try the old branch.. but .. its old).

AsciiWolf, 2022-03-09:
There is an [dump1090 fork](https://github.com/SDRplay/dump1090) with SDRplay support added, but it is not ideal to use a fork.

bkoehm, 2022-12-18:
I have built on work by @mutability ( January 2019, soapysdr branch) to add SoapySDR support, including support for SDRPlay devices.  It's been opened as pull request #209.  I'm happy to work with code reviewers on this.

The only device I have available to me right now is the SDRPlay RSPdx and I have tested with this device.

These are things that I think need particular review:
1. I'm unsure of why Modes.gain was divided by 10 in the Jan 2019 work.  Still needed?  Is that needed for RTL-SDR devices only?  If so, anybody know what SoapySDRDevice_getHardwareKey() returns for RTL-SDR?  (This is printed to stdout as `soapy: hardware key is ...`.)
2. The FIFO work probably needs to be reviewed.  I copied what I saw in other sdr_*.c implementations.
3. Similarly, it would be helpful to review if outbuf->sampleTimestamp is being calculcated correctly.  I am not fully confident that it's going to be correct for all devices.

Additionally, I don't have a RSPduo, but I have noticed in [SoapySDRPlay3](https://github.com/pothosware/SoapySDRPlay3) code that it appears unlikely that RSPduo can support the 2.4 MHz sample rate.  Someone can correct me if I am wrong, but it seems dump1090 only demodulates a 2.4 MHz sample rate, so RSPduo may not be supportable.

mutability, 2022-12-23:
Thanks for picking this up! I'll leave some comments directly on the PR

rcarsey1, 2023-02-22:
> Additionally, I don't have a RSPduo, but I have noticed in [SoapySDRPlay3](https://github.com/pothosware/SoapySDRPlay3) code that it appears unlikely that RSPduo can support the 2.4 MHz sample rate. [...]

I have the RSPduo.  If operated in single-tuner mode (whether you use Tuner 1 or Tuner 2), then the sample rate can be 2-10Mhz and the IF bandwidth can be 200khz-8Mhz.  

If operated in dual-tuner mode, then:  the first application that uses the device is designated "master" (again, Tuner 1 or Tuner 2 can be chosen).  The sample rate can be 6Mhz or 8Mhz, and the IF bandwidth can be 200khz-1.536Mhz.  The second application which uses the device, MUST use the same sample rate as the master.  

The RSP people have their own dump1090 which seems to be working OK in dual-tuner mode for me..  However, they forked it years ago and is missing a lot of the newer features (and even some basic features [e.g. very terse output in the JSON output file]).  Let me know if I can assist.

bkoehm, 2023-02-22:
@rcarsey1 Thanks.  If you're up for it, you can try out #209 and see if it works with your RSPduo in single-tuner mode.

rcarsey1, 2023-02-22:
i just tried #209 with the RSPduo (the RSP1A worked OK too as expected).  In single-tuner mode, it looks like it works as expected for the "Tuner 1 50 ohm" antenna.  For the "Tuner 2 50 ohm" antenna, I was receiving no data.  I have a VHF antenna hooked up to Tuner 2 at the moment, so I can't say for certain if its a software problem (I'll be at the airport soon to swap antennas and check this out).  Even with a VHF antenna though, I'd probably expect some traffic to show up.

In dual-tuner mode,  as expected, it failed because 2.0Mhz is not a valid sample rate:

[RSPduo_output.txt](https://github.com/flightaware/dump1090/files/10800198/RSPduo_output.txt)

rcarsey1, 2023-02-22:
typo in my posting here.  I'm a bit new to SoapySDR, so it might take me absurdly long to figure out what is going on.  I can offer you a SSH login into my machine if you'd like to spend some time looking at it.  As far as I can tell, it shouldn't make any difference which tuner/antenna is selected while in single-tuner mode (thats what the [datasheet ](https://www.sdrplay.com/wp-content/uploads/2018/05/RSPduoDatasheetV0.6.pdf)indicates; and RSP's own dump1090 works OK on either tuner).

In the meantime, if you'd like that login to mess around with the RSP1A and RSPduo, shoot me a line at rcarsey2@gmail.com.

mutability, 2023-08-08:
#209 has been merged (with some tweaks) to the dev branch.

Is there anything else in this issue that still needs looking at?

bkoehm, 2023-08-08:
From my perspective this can be closed now but if @rcarsey1 is available to test with the RSP1A and the RSPduo (in single tuner mode), that might be a good additional test.

mutability, 2023-11-10:
soapysdr support is merged/released now, so I'll close this.

commit 0c71b3db (2016-10-01): Fix up unsigned subtractions & use of abs.
Also fix heading difference which was just wrong, signed or not.

Should fix #150
closed 2023-11-10 by mutability