flightaware/dump1090 #153: Adaptive Gain - Burst Mode Configuration Questions

VirusPilot, 2021-09-06:
I have been testing the burst mode since it has been released in the following application which certainly is only an edge case for dump1090: [https://github.com/b3nn0/ogn2dump1090](url)

Here the requirement is not to achieve maximum range rather than reliable reception of nearby traffic, even if airplanes are taxiing close to the receiver location as this application is intended to support the local air traffic controller. With burst mode enabled I was hoping to be able to set the receiver up with some moderate gain (e.g. 40dB) and let the burst mode do the job to decrease the gain as soon as the traffic approaches the traffic pattern and then will touch down on the runway close to the receiver site. This is somehow similar to our Stratux project requirements but the ownship transponder is "killing" the burst mode there as discussed [https://github.com/flightaware/dump1090/pull/144#issuecomment-891684768](url)).

Unfortunately the default burst mode config takes 14s per gain step so that it takes >1min to decrease the gain down to a level which prevents too many loud messages. In many cases this is way too slow so I played around with some of the burst mode parameters but no success, still 14s per gain step.

So how do I have to change the parameters so that a gain step only requires a few seconds? 

As a suggestion it might be nice to have a few burst **modes** instead, e.g. fast, moderate, slow.

mutability, 2021-09-23:
Try changing `--adaptive-burst-change-delay`, `--adaptive-burst-loud-runlength`, `--adaptive-burst-quiet-runlength`

VirusPilot, 2021-09-23:
Did the following changes and still 14 seconds for each gain step:

ADAPTIVE_BURST=yes
ADAPTIVE_BURST_ALPHA=1
ADAPTIVE_BURST_CHANGE_DELAY=1
ADAPTIVE_BURST_LOUD_RUNLENGTH=1
ADAPTIVE_BURST_QUIET_RUNLENGTH=1

mutability, 2021-09-23:
I assume this is in `/etc/default/dump1090-fa`. You can't just make up new settings in there and expect it to do anything; either you also have to teach `start-dump1090-fa` about them, or pass the new command-line switches in `EXTRA_OPTIONS`

VirusPilot, 2021-09-23:
Alright, now I am on board, will test.

VirusPilot, 2021-09-27:
Testing concluded, it works quite well with the following parameter set:

`EXTRA_OPTIONS="--adaptive-burst-alpha 0.7 --adaptive-burst-change-delay 1 --adaptive-burst-loud-runlength 1 --adaptive-burst-quiet-runlength 5"`

I have modified `debian/dump1090-fa.default` accordingly in my fork.

Thanks for your support!

mutability, 2021-10-01:
:+1: The burst mode is quite experimental so knowing what settings work for you is very useful, thanks for the feedback!

commit 2b47fe5b (2016-10-14): Fix knots -> km/h conversion in interactive mode.

Fixes #153
closed 2021-10-01 by mutability