flightaware/dump1090 #158: CPUFEATURES cross-compile.

milkylainen, 2021-09-22:
Having issues with cross compiling.

Makefile.cpufeatures:
CPUFEATURES_UNAME := $(shell uname)
CPUFEATURES_ARCH := $(shell uname -m)

Immediate set instead of lazy set?
ARCH=aarch64 breaks as uname -m is a x86_64 host.

mutability, 2021-09-22:
Yes, `:=` because we don't want to call `uname` every time that those variables are referenced.

You should be able to override on the command line, no?

milkylainen, 2021-09-22:
For whatever reason that didn't work.
I patched it to a lazy set ?= and handed the ARCH from the build environment on the commandline.

mutability, 2021-09-22:
I can't reproduce this, setting CPUFEATURES_ARCH on the command line works fine here:

```
$ uname -m
x86_64
$ make ARCH=aarch64 CPUFEATURES_ARCH=aarch64 CC=aarch64-linux-gnu-gcc
[...]
aarch64-linux-gnu-gcc -Icpu_features/include -std=c99 -O -g -DSTACK_LINE_READER_BUFFER_SIZE=1024 -DNDEBUG -DHAVE_STRONG_GETAUXVAL -c cpu_features/src/cpuinfo_aarch64.c -o cpu_features/src/cpuinfo_aarch64.o
[...]
```

mutability, 2021-09-22:
oh, I guess you probably tried with env vars. Try on the command line like I did instead.

All that said, I forget why CPUFEATURES_ARCH does its own lookup, having it default to the value of ARCH would be reasonable

mutability, 2021-09-23:
Try that commit?

milkylainen, 2021-09-23:
Ah. Yes. I pushed them on env. But the suggested solution works nicely for me.
Built and tested.

As a side note I'm using dump1090 with librtlsdr 0.8.0 under an qemu emulated aarch64 on an embedded x86_64 AMD R1000 with the footprint of a RPi. :) Works ok, even with instruction emulation.

Thanks for the help!

commit 059e48b8 (2021-09-23): Allow env vars (not only command line overrides) to set CPUFEATURES_{ARCH,UNAME}.
Change CPUFEATURES_{ARCH,UNAME} defaults to use {ARCH,UNAME} values.

Fixes #158
closed 2021-12-07 by mutability