flightaware/dump1090 #160: Unmet dependencies on Debian Bullseye (11)

sv1, 2021-09-28:
Trying to compile latest code from github

$ dpkg-buildpackage -b
dpkg-buildpackage: info: source package dump1090-fa
dpkg-buildpackage: info: source version 7.0~dev
dpkg-buildpackage: info: source distribution UNRELEASED
dpkg-buildpackage: info: source changed by Oliver Jowett <oliver.jowett@flightaware.com>
dpkg-buildpackage: info: host architecture i386
dpkg-source --before-build .
dpkg-checkbuilddeps: error: Unmet build dependencies: dh-systemd
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)

makrsmark, 2021-09-28:
dh-systemd has been rolled into debhelper for bullseye. Not a permanent fix, but you can get it building by changing the dependencies.

```
sed -i  's/, dh-systemd/debhelper/' ./debian/control
```

airvb, 2021-10-11:
Hello, 

Can you describe how to use yr fix ? 

Thanks

> dh-systemd has been rolled into debhelper for bullseye. Not a permanent fix, but you can get it building by changing the dependencies.
> 
> ```
> sed -i  's/, dh-systemd/debhelper/' ./debian/control
> ```

mutability, 2021-12-07:
dev branch has been updated to support bullseye

closed 2021-12-07 by mutability