flightaware/dump1090 #166: Error compiling for Mac M1 processor

devleaks, 2021-12-06:
Hello,
I successfully installed brew required packages then make fails at:
`
dump1090 % make                             
Building with:
  Version string:  unknown
  DSP mix:         arm
  RTLSDR support:  yes
  BladeRF support: yes
  HackRF support:  yes
  LimeSDR support: no
cc -Icpu_features/include -std=c99 -O -g -DSTACK_LINE_READER_BUFFER_SIZE=1024 -DNDEBUG -DHAVE_SYSCTLBYNAME -c cpu_features/src/cpuinfo_arm.c -o cpu_features/src/cpuinfo_arm.o
In file included from cpu_features/src/cpuinfo_arm.c:15:
cpu_features/include/cpuinfo_arm.h:118:2: error: "Including cpuinfo_arm.h from a non-arm target."
#error "Including cpuinfo_arm.h from a non-arm target."
 ^
1 error generated.
make: *** [cpu_features/src/cpuinfo_arm.o] Error 1
`
Thank you for your input.

mutability, 2021-12-06:
Try building with ARCH=aarch64 for now (apple apparently named it
differently to linux et al)

mutability, 2021-12-07:
bc72177c8b8d3c540657548716f6024ce2bceabd should fix this in theory

devleaks, 2021-12-07:
Makefile needed a lot more twisting than suggested:
- libusb-1.0 not properly referenced (no -Lpath)
- "arm64" partially matches "arm", etc.
- cpu_features compilation is delegated to Makefile.cpufeatures
- etc.

when I finally got the linker to link, I bumped into this:

```
Undefined symbols for architecture arm64:
  "_CpuFeatures_GetHardwareCapabilities", referenced from:
      _GetAarch64Info in cpuinfo_aarch64.o
  "_CpuFeatures_IsHwCapsSet", referenced from:
      _GetAarch64Info in cpuinfo_aarch64.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [dump1090] Error 1
```

Googling for "Including cpuinfo_arm.h from a non-arm target." led me to more editing in google's cpu_features subrepo.
See [here](https://github.com/google/cpu_features/issues/179).

P.

devleaks, 2021-12-07:
please note that dump1090 compiles without cpu_features.

mutability, 2021-12-07:
I disabled cpufeatures for that arch/os in fa8a066b4c7599d61d214ab11fd7c4d0d09ff40d

For the other issues I'll probably need a PR as I don't have an environment to test any of this.