flightaware/dump1090 #168: Compilation Failure ubuntu Impish

wx4cb, 2022-01-11:
followed all the directions as per buster, changed the debian.control file to have debhelper instead of dh-systemd.

seems to compile ok for the most part, but i get the following errors art the end of the compile

i have rtl-sdr compiled from source so it appears it's looking for the rtl package which isnt obviously installed ?

```
dh_shlibdeps: warning: Compatibility levels before 10 are deprecated (level 9 in use)
        install -d debian/dump1090-fa/DEBIAN
        dpkg-shlibdeps -Tdebian/dump1090-fa.substvars debian/dump1090-fa/usr/bin/view1090-fa debian/dump1090-fa/usr/bin/dump1090-fa debian/dump1090-fa/usr/lib/dump1090-fa/starch-benchmark
dpkg-shlibdeps: error: no dependency information found for /usr/local/lib/librtlsdr.so.0 (used by debian/dump1090-fa/usr/bin/dump1090-fa)
Hint: check if the library actually comes from a package.
dh_shlibdeps: error: dpkg-shlibdeps -Tdebian/dump1090-fa.substvars debian/dump1090-fa/usr/bin/view1090-fa debian/dump1090-fa/usr/bin/dump1090-fa debian/dump1090-fa/usr/lib/dump1090-fa/starch-benchmark returned exit code 255
dh_shlibdeps: error: Aborting due to earlier error
make: *** [debian/rules:57: binary] Error 2
dpkg-buildpackage: error: fakeroot debian/rules binary subprocess returned exit status 2
```

mutability, 2022-01-12:
Yeah, you need a correctly packaged version of librtlsdr installed to do a package build of dump1090. If you also have a compiled-from-source version present, that will interfere with the build.

closed 2022-01-12 by mutability

wx4cb, 2022-01-12:
i found a work aroudn for this. basically to get shlib to ignore failed dependencies and got it installed and running

mutability, 2022-01-12:
Sure. Be aware that the error is intentional; if you disable the check then you'll end up with a package that has linked against a library that the package management system knows nothing about, and so you'll run into unexpected runtime problems if that library is ever changed or removed (or if you try to install the built package on a different system)