flightaware/dump1090 #175: Country flag of Switzerland is squeezed

by, 2022-03-06:
The original country flag from https://dynamospanish.com/flags/countries/switzerland/ is square (as it should be); at https://github.com/flightaware/dump1090/blob/master/public_html/flags-tiny/Switzerland.png, it is stretched to the rectangular format of the other flags

mutability, 2022-03-07:
The original is actually not square (see the topmost "tiny" flag, which is what we use):

[image]

If you have a better version with the correct aspect ratio (it would need to still be a rectangular image of the same size, with transparency at the left/right edges), could you provide the new image or a PR updating it?

by, 2022-03-07:
Yes, the original is incorrect; what about this one:
[image]

commit 0dede580 (2022-10-10): Merge pull request #205 from flightaware/square_swiss_flag

Fix Swiss flag aspect ratio (fixes #175)
commit 679a5668 (2022-10-10): Fix Swiss flag aspect ratio (fixes #175)
closed 2022-10-10 by mutability