flightaware/dump1090 #178: dropped_samples counting is broken

wiedehopf, 2022-04-27:
https://github.com/flightaware/dump1090/blob/849a3b73299b4f56620ab16a6b62d88e17f35608/dump1090.c#L897
https://github.com/flightaware/dump1090/blob/849a3b73299b4f56620ab16a6b62d88e17f35608/sdr_rtlsdr.c#L335

This fix for readsb-protobuf also applies to dump1090-fa:
https://github.com/Mictronics/readsb-protobuf/pull/21

actually two issues:
- outbuf->dropped is not reset and in dump1090.c dropped is used regardless of the magbuf discontinuous flag
- static variable dropped is never reset in sdr_rtl.c

mutability, 2022-05-09:
Thanks for the fixes. I might move the initialization of `outbuf->dropped` into the fifo code so it's always done

mutability, 2022-06-09:
nb: static variable seems to be correctly reset already

wiedehopf, 2022-06-09:
yeah seems this wasn't correctly patched when Mictronics used your fifo code.
(i don't have any of the fifo code in my fork, still have the old dump1090 code for managing the buffers)

wiedehopf, 2022-06-09:
Pretty sure the other part of the patch was still necessary.
The static dropped would have been reset and outbuf->dropped would have been added each time the buffer was being reused.

closed 2022-06-09 by wiedehopf

mutability, 2022-06-09:
which "other part" do you mean? The change in the linked PR resets the per-buffer `dropped` on each acquire. I don't think there's anything else?

wiedehopf, 2022-06-09:
Yeah you fixed that by setting outbuf-dropped to zero in fifo_acquire.
Sorry if the comment was ambiguous.
I meant you already did the part of the fix necessary for dump1090-fa

PR #184 (2022-06-09): Fix dropped samples counting
Zero out `dropped` on FIFO buffers for contiguous buffers; document that dropped == 0 for contiguous buffers.

Fixes #178