flightaware/dump1090 #179: cannot view previously captured files interactively with `ifile` directive

jamesleytevidal, 2022-04-28:
Hi there, 

in previous versions of dump1090 (non-fa), I was able to capture signals with the rtl_sdr utility to a file, and play those back with dump1090 using --ifile. When I do this with this version of dump1090, the interactive interface comes up but no flight data is populated. 

I tried the --iformat UC8 argument with it as well with no difference. I can use --interactive live with an rtl-sdr dongle and get data, so this seems to be an issue only when supplying an input file. 

What am I missing? is there another argument I need to supply?

mutability, 2022-05-09:
How did you capture the data?
What's the command line you're trying to use to read the data?

jamesleytevidal, 2022-05-09:
capture was originally performed with the rtl_sdr utility.

Looking back through some old tickets, I think we've figured out what's going on - sample rate in the original captures was 2000000, new captures at sample rate of 2400000 render fine. So we have a way to deal with it, we just need to recapture the data. Closing. 

Thanks!

closed 2022-05-09 by jamesleytevidal

mutability, 2022-05-09:
Ah yep that'd do it - the 2MHz demodulator was replaced with the 2.4MHz one quite a while ago.