flightaware/dump1090 #191: FEATURE REQ: Distance filter

robsymax, 2022-08-07:
It would be great to be able to limit the aircraft displayed to those within a set distance to the receiver for the purpose of gauging rough aircraft noise. Is this easy to implement?

Bdavy92960, 2023-02-01:
I would also like to see this feature, and an associated audible alarm option for aircraft within a chosen distance and below a chosen altitude. This would be a very useful option for UAV pilots as a warning of nearby aircraft.

JonMurphy, 2024-06-03:
+1 for me!!

oam7575, 2024-10-16:
@robsymax  @Bdavy92960  @JonMurphy 
Hello,

I have created a dev branch that implements a distance slider to filter the display of aircraft.
Pull request #249 is open however the raw files can be downloaded from the public_html directory of my dev branch here : https://github.com/oam7575/dump1090/tree/dev-distance-filter

If you would like to manually upgrade your installation do the following.
**Backup** 
Backup these files from your current installation
- index.html
- config.js
- planeObject.js
- script.js
- style.css

**Upgrade**
Replace the listed files with the versions in my dev branch.

-

oam7575, 2024-10-18:
Hello, @robsymax @Bdavy92960 @JonMurphy

Distance and Altitude alert beeps have been implemented in my pull request #250.
#250 also includes the distance slider previously implemented in #249.

As per my previous comment, you can manually upgrade your installation by downloading the raw files from the public_html folder of my branch : https://github.com/oam7575/dump1090/tree/dev-distance_alert_altitude_alert

**Note :** The changes for the alert beeps are in a separate branch