flightaware/dump1090 #211: Request: Missing some statistics in the json files

misko85, 2022-12-27:
I tried to find the following statistics in the JSON files. 
- Total Aircraft
- With Positions
- Position History
- ADS-B Message Rate

Are they not available? 
Could these statistics still be added? 
Best Regards.

mutability, 2022-12-28:
> Total Aircraft
> With Positions

These can be derived from aircraft.json

> Position History

dump1090 doesn't maintain position history, this is a web-interface-side thing.

> ADS-B Message Rate

There's lots of message counters in existing stats output; differentiate those wrt time.

misko85, 2022-12-29:
Sorry, but I can´t see any payload in aircraft.json that will give me the number of "Total Aircraft" and "With Position". There ist only one object with a number of total messages and a array with aircraft informations. The only option ist to count the array length to get the number of airplanes. But these are all airplains with and without positions.

Output of the aircraft.json:
{"now":1672333118.1,"messages":29352292,"aircraft":[{"hex":"4853d3","alt_baro":29775,"alt_geom":28950,"gs":420.6,"track":147.8,"baro_rate":1920,"version":0,"nac_p":8,"nac_v":1,"sil":2,"sil_type":"unknown","mlat":[],"tisb":[],"messages":18,"seen":13.2,"rssi":-21.3},{"hex":"49d034","flight":"TVS434F ","alt_baro":40000,"alt_geom":38925,"gs":388.3,"ias":229,"tas":434,"mach":0.76,"track":205,"track_rate":0,"roll":-0.5,"mag_heading":211.5,"baro_rate":0,"geom_rate":0,"squawk":"3517","emergency":"none","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":40000,"nav_altitude_fms":40000,"nav_heading":212.3,"lat":51.783965,"lon":5.28717,"nic":8,"rc":186,"seen_pos":24.9,"version":2,"nic_baro":1,"nac_p":8,"nac_v":1,"sil":3,"sil_type":"perhour","gva":1,"sda":2,"modea":true,"modec":true,"mlat":[],"tisb":[],"messages":343,"seen":5.2,"rssi":-20.9},{"hex":"c038a3","flight":"ACA843  ","alt_baro":28925,"alt_geom":28375,"gs":427.8,"ias":313,"tas":470,"mach":0.808,"track":305.1,"track_rate":0,"roll":-0.2,"mag_heading":296.2,"baro_rate":1088,"geom_rate":1152,"squawk":"2531","emergency":"none","category":"A5","nav_qnh":1012.8,"nav_altitude_mcp":32000,"nav_heading":295.3,"nav_modes":["autopilot","vnav","lnav","tcas"],"lat":50.974654,"lon":6.820831,"nic":8,"rc":186,"seen_pos":0.2,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"modec":true,"mlat":[],"tisb":[],"messages":890,"seen":0.1,"rssi":-14.7},{"hex":"4073ce","flight":"VIR365  ","alt_baro":40000,"alt_geom":39150,"gs":431.9,"ias":255,"tas":480,"mach":0.84,"track":292.2,"track_rate":0,"roll":-0.2,"mag_heading":284.2,"baro_rate":160,"geom_rate":128,"squawk":"3260","emergency":"none","category":"A5","nav_qnh":1012.8,"nav_altitude_mcp":40000,"nav_heading":286.9,"nav_modes":["autopilot","vnav","lnav","tcas"],"lat":50.60177,"lon":6.547852,"nic":8,"rc":186,"seen_pos":0.1,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"modec":true,"mlat":[],"tisb":[],"messages":751,"seen":0.1,"rssi":-19.9},{"hex":"3c65cd","flight":"DLH3U   ","alt_baro":38975,"alt_geom":38100,"gs":491.8,"ias":243,"tas":448,"mach":0.784,"track":105.4,"track_rate":0,"roll":-0.2,"mag_heading":108.6,"baro_rate":-64,"geom_rate":-32,"squawk":"1165","category":"A3","nav_qnh":1012.8,"nav_altitude_mcp":24992,"lat":50.565506,"lon":5.270171,"nic":8,"rc":186,"seen_pos":10,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"mlat":[],"tisb":[],"messages":233,"seen":10,"rssi":-18.5},{"hex":"485064","flight":"KLM31W  ","alt_baro":25000,"alt_geom":24425,"gs":379,"ias":254,"tas":364,"mach":0.612,"track":131.5,"track_rate":0,"roll":0.2,"mag_heading":143.8,"baro_rate":-64,"geom_rate":-32,"squawk":"1000","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":25024,"nav_altitude_fms":25008,"nav_modes":["autopilot","vnav","tcas"],"lat":51.452792,"lon":6.669159,"nic":8,"rc":186,"seen_pos":0.3,"version":2,"nic_baro":1,"nac_p":10,"nac_v":2,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"modec":true,"mlat":[],"tisb":[],"messages":3625,"seen":0,"rssi":-5.6},{"hex":"40666c","flight":"EZY28QP ","alt_baro":36025,"alt_geom":35300,"gs":386.5,"ias":255,"mach":0.772,"track":308.4,"mag_heading":266.3,"baro_rate":-64,"geom_rate":32,"squawk":"3554","emergency":"none","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":36000,"nav_heading":0,"lat":50.212501,"lon":5.572963,"nic":8,"rc":186,"seen_pos":61.3,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","modec":true,"mlat":[],"tisb":[],"messages":211,"seen":0.3,"rssi":-14.8},{"hex":"4caca0","flight":"SAS59J  ","alt_baro":2250,"alt_geom":2200,"gs":131.4,"ias":164,"tas":168,"mach":0.256,"track":233,"track_rate":0.09,"roll":0.7,"mag_heading":232.4,"baro_rate":-736,"geom_rate":-832,"squawk":"5663","emergency":"none","category":"A3","nav_qnh":1007.2,"nav_altitude_mcp":4000,"nav_heading":232.7,"lat":51.352249,"lon":6.905056,"nic":8,"rc":186,"seen_pos":0.2,"version":2,"nic_baro":1,"nac_p":11,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"mlat":[],"tisb":[],"messages":1687,"seen":0.1,"rssi":-5.4},{"hex":"3c49b1","flight":"EWG46T  ","alt_baro":20850,"alt_geom":20425,"gs":341.8,"ias":299,"tas":400,"mach":0.66,"track":215,"track_rate":-1.12,"roll":-20.9,"mag_heading":217.4,"baro_rate":2624,"geom_rate":2592,"squawk":"4113","emergency":"none","category":"A3","nav_qnh":1013.6,"nav_altitude_mcp":35008,"nav_heading":170.2,"lat":50.791992,"lon":6.280394,"nic":8,"rc":186,"seen_pos":0.8,"version":2,"nic_baro":1,"nac_p":9,"nac_v":1,"sil":3,"sil_type":"perhour","gva":2,"sda":2,"modea":true,"modec":true,"mlat":[],"tisb":[],"messages":2145,"seen":0.2,"rssi":-15.8},{"hex":"43c6f4","flight":"RRR2303 ","alt_baro":35975,"alt_geom":35275,"gs":409.7,"ias":279,"tas":480,"mach":0.836,"track":286.9,"track_rate":-0.03,"roll":0,"mag_heading":279.7,"baro_rate":0,"geom_rate":0,"squawk":"3552","emergency":"none","category":"A5","nav_qnh":1013.6,"nav_altitude_mcp":36000,"nav_heading":312.9,"lat":50.728998,"lon":6.191483,"nic":8,"rc":186,"seen_pos":0 [... cut ...] 4.5,"rssi":-18.9},{"hex":"495210","alt_baro":900,"alt_geom":900,"gs":111.2,"track":232.3,"baro_rate":-832,"squawk":"1000","category":"A3","nav_altitude_fms":3008,"lat":51.316859,"lon":6.830139,"nic":8,"rc":186,"seen_pos":68.5,"version":2,"nac_v":2,"sil_type":"perhour","modea":true,"mlat":[],"tisb":[],"messages":3876,"seen":46.4,"rssi":-18.6},{"hex":"4ca95e","category":"A3","version":2,"sil_type":"perhour","modea":true,"modec":true,"mlat":[],"tisb":[],"messages":16491,"seen":192.5,"rssi":-21.7}]}

mutability, 2022-12-29:
Yes, that's exactly what you do: iterate through the aircraft array and classify the aircraft however you want.

Here's an old example of doing just that: https://github.com/mutability/dump1090-tools/blob/master/collectd/dump1090.py#L199

(graphs1090 probably has more up to date code)

q-wertz, 2025-05-28:
Maybe stupid question:
Is there a reason why the time value is not stored in the json output?