flightaware/dump1090 #216: Cannot compile on macOS Ventura

JimMerkel, 2023-02-05:
I get the following from: make RTLSDR=yes
``
make: pkg-config: Command not found
make: pkg-config: Command not found
Building with:
  Version string:  unknown
  Architecture:    x86_64
  DSP mix:         x86
  RTLSDR support:  yes
  BladeRF support: no
  HackRF support:  no
  LimeSDR support: no
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c dump1090.c -o dump1090.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c anet.c -o anet.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c interactive.c -o interactive.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c mode_ac.c -o mode_ac.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c mode_s.c -o mode_s.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c comm_b.c -o comm_b.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c net_io.c -o net_io.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c crc.c -o crc.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c demod_2400.c -o demod_2400.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c stats.c -o stats.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c cpr.c -o cpr.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c icao_filter.c -o icao_filter.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c track.c -o track.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c util.c -o util.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c convert.c -o convert.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c ais_charset.c -o ais_charset.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c adaptive.c -o adaptive.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c cpu.c -o cpu.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c sdr.c -o sdr.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c fifo.c -o fifo.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c sdr_ifile.c -o sdr_ifile.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c dsp/helpers/tables.c -o dsp/helpers/tables.o
cc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -DMISSING_NANOSLEEP -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DSTARCH_MIX_X86 -O3 -g -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -W  -c sdr_rtlsdr.c -o sdr_rtlsdr.o
sdr_rtlsdr.c:54:10: fatal error: 'rtl-sdr.h' file not found
#include <rtl-sdr.h>
         ^~~~~~~~~~~
1 error generated.
make: *** [sdr_rtlsdr.o] Error 1
``

mutability, 2023-02-06:
> fatal error: 'rtl-sdr.h' file not found #include <rtl-sdr.h>

You don't have librtlsdr installed correctly.

JimMerkel, 2023-02-06:
Ok, I did a git clone of librtlsdr and its in a directory in dump1090.  Do I set RTLSDR_PREFIX = librtlsdr in the Makefile ?