flightaware/dump1090 #219: Query arguments no longer supported

raccettura, 2023-06-11:
Seems like in a somewhat recent version query arguments notably (?banner=hide&sidebar=hide) no longer work.

Quickly tracing this I found:

`applyUrlQueryStrings` in script.js does fire, however something in `resetMap()` seems to cause it break and thus it never gets beyond that point.... didn't get any deeper than that.

eric1tran, 2023-06-12:
Working OK here. What's the full url you're using? Do you have console log output?

raccettura, 2023-06-12:
Using:
```
/skyaware/?banner=hide&sidebar=hide
```

Which has been the case for maybe a year with no issue.  Most of the console is boring minus maybe:
/skyaware/data-978/receiver.json?_=1686599433516 404
skyaware/upintheair.json 404
and
[image]

mutability, 2023-06-13:
At a glance this is probably due to a missing receiver location.
(It shouldn't break in that case, which is a bug, but that is possibly what changed in your setup)

raccettura, 2023-06-13:
I've never had a 978mhz antenna, so no change in my setup in a long time.

mutability, 2023-06-14:
Okay. Do you get a receiver location dot and range circles when you load the page without a query string?

raccettura, 2023-06-14:
Negative.  But everything else functions.  Only error is:
> Error reading SkyAware978 receiver.json. SkyAware978 may be disabled

Which is correct/expected.

mutability, 2023-06-15:
Okay, so as I mentioned above, the trigger for this is likely that you do not have a receiver location set.