flightaware/dump1090 #231: dev branch soapy sdr support

rush0815, 2023-11-04:
Hi folks,
finally soapy sdr support has been included in the dev branch. But from where can I get all the parameters that are possible for soapy driven devices:

--device <string>          select/configure device
--channel <num>            select channel if device supports multiple channels (default: 0)
--antenna <string>         select antenna (default depends on device)
--bandwidth <hz>           set the baseband filter width (default: 3MHz, SDRPlay: 5MHz)
--enable-agc               enable Automatic Gain Control if supported by device
--gain-element <name>:<db> set gain in dB for a named gain element

To open a device I need to put in 
--device device_id=<whatever SoapySDRUtil --find put out>
What else can I put into --device ? What format should/can it have?

From where can I get <string> for antenna names?
From where can I get <string>:<db> for gain settings?

I testet `dump1090 --device-type soapy --device device_id=1 --bandwidth 1800000 --enable-agc` for my rtl-sdr stick.
But that seems to no work because the startup gives:

soapy: driver key:      RTLSDR
soapy: hardware key:    R820T
[R82XX] PLL not locked!
soapy: total gain:      49.6dB; TUNER=49.6dB
soapy: frequency:       1090.0 MHz
soapy: sample rate:     2.4 MHz
soapy: bandwidth:       0.0 MHz
soapy: AGC mode:        manual
soapy: antenna:         RX

So it looks like --bandwith and --enable-agc has no impact.

The same with a LimeSDR mini look so:
`
...
[INFO] RX LPF configured
[INFO] Selected RX path: LNAW
[WARNING] Rx LPF min bandwidth is 4MHz when TIA gain is set to -12 dB
[INFO] RX LPF configured
soapy: total gain:      0.0dB; TIA=0.0dB; LNA=12.0dB; PGA=-12.0dB
soapy: frequency:       1090.0 MHz
soapy: sample rate:     2.4 MHz
soapy: bandwidth:       1.8 MHz
soapy: antenna:         LNAW
soapy: DC offset mode:  automatic
soapy: DC offset:  I=0.000 Q=0.000
...
`
Here I can also see how the gain elements are named. Well, but I have to start the program first to get it out.

Maybe @mutability could post some example calls for all that options ?

rush0815, 2023-11-04:
Btw: --gain-element TIA:<gain> does not work on a LimeSDR mini. I will set TIA to 0

mutability, 2023-11-10:
Most of your questions are probably better addressed by the SoapySDR or individual driver documentation.

> What else can I put into --device ? What format should/can it have?

See https://github.com/pothosware/SoapySDR/wiki/DriverGuide#registration-api
It's basically just a driver-specific list of key/value args.

> From where can I get for antenna names?
> From where can I get : for gain settings?

See `SoapySDRUtil --probe=<device>` output

> So it looks like --bandwith and --enable-agc has no impact.

`--bandwidth` works fine here, perhaps your driver is old.

`--enable-agc` handling looks like a general bug/oversight in SDR-driver-specific option handling (the option is being consumed early by the rtlsdr driver and never gets seen by the soapysdr driver)

mutability, 2023-11-10:
1b4fd182c4e991e42e9f3b814454eb1204356887 and 9160dba3710655bcb50bbbd4c48ad07b2f3163d2 on the dev branch should fix `--enable-agc`