flightaware/dump1090 #242: Add devscripts for Debian/Raspbian for list of dependencies.

apatheticposture, 2024-06-17:
dch is not included by default on minimal Raspbian installs; thus the "Building under bullseye, buster, or stretch" section should have devscripts added to the apt-get command:

`sudo apt-get install build-essential fakeroot debhelper librtlsdr-dev pkg-config libncurses5-dev libbladerf-dev libhackrf-dev liblimesuite-dev libsoapysdr-dev devscripts`

c3kkos, 2024-09-26:
I thought i missed that.

But after that apt install command, i still get this:

```
 dpkg-shlibdeps: error: no dependency information found for /usr/local/lib/arm-linux-gnueabihf/librtlsdr.so.0 (used by debian/dump1090-fa/usr/bin/dump1090-fa)
Hint: check if the library actually comes from a package.
dh_shlibdeps: error: dpkg-shlibdeps -Tdebian/dump1090-fa.substvars debian/dump1090-fa/usr/lib/dump1090-fa/starch-benchmark debian/dump1090-fa/usr/bin/view1090-fa debian/dump1090-fa/usr/bin/dump1090-fa returned exit code 255
dh_shlibdeps: error: Aborting due to earlier error
make: *** [debian/rules:69: binary] Error 2
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2
```

after this command: dpkg-buildpackage -b --no-sign
bullseye as target

lstolcman, 2025-04-16:
closed via https://github.com/flightaware/dump1090/pull/234 in version 10.0