flightaware/dump1090 #252: Running flightaware/dump1090 on Raspberry Pi but no flights found

rosen1710, 2024-12-08:
I am running dump1090 on Raspberry Pi 4 (model B with 8 GB RAM) with Ubuntu desktop OS. The program starts correctly but no flights are found. I have used the same antenna, SDR and placement with another Windows PC and with https://github.com/gvanem/Dump1090 installed (it is Windows compatible) so that everything works fine with many flights detected.

I start the flightaware/dump1090 program with this command: `./dump1090 --interactive --net`
Few seconds after starting the program I see this:
[image]

The problem is that there are a lot of flights around me, but none of them has been detected by dump1090 on the Raspberry Pi.

After program stop, the log after execution is:
```
Sun Dec  8 11:22:23 2024 EET  dump1090-fa unknown starting up.
rtlsdr: using device #0: Generic RTL2832U OEM (RTLSDRBlog, Blog V4, SN 00000001)
Found Rafael Micro R828D tuner
rtlsdr: tuner gain set to 49.6 dB (gain step 28)
[R82XX] PLL not locked!
[R82XX] PLL not locked!
Sun Dec  8 11:32:55 2024 EET  Normal exit.
```

I don't know what the problem is - is it software problem, driver problem or hardware problem. At first I thought that the power consumption of the antenna and the SDR was too high, so I bought a powered USB 2.0 hub, which I am using now, but nothing has changed. Can anyone help me fix this problem and start receiving flight data with dump1090 on Raspberry Pi with Ubuntu desktop OS?

mutability, 2024-12-08:
The rtlsdr v4 requires a replacement librtlsdr to work correctly (affects everything, not only dump1090) -- see https://www.rtl-sdr.com/V4/

closed 2024-12-08 by mutability

rosen1710, 2024-12-08:
> The rtlsdrblog v4 requires a replacement librtlsdr to work correctly (affects everything, not only dump1090) -- see https://www.rtl-sdr.com/V4/

Thank you - this solved my issue.

klubi, 2025-05-04:
> > The rtlsdrblog v4 requires a replacement librtlsdr to work correctly (affects everything, not only dump1090) -- see https://www.rtl-sdr.com/V4/
> 
> 
> 
> Thank you - this solved my issue.

Can you share which part solved issue is your case?
I'm running same setup but with DietPi. Managed to build all libs from source (except piaware).
I was able to claim my feeder, but not a single plane was found in last 24h.
When I run dump1090-fa on my Mac using rtl, I can see a plane every couple of minutes.
I can see that PLL message in my logs as well, but have no idea how to fix it:/