flightaware/dump1090 #257: Compilation will fail with GCC-15

ConiKost, 2025-03-01:
Since dump1090 enables by default `-Werror`, starting with GCC-15, compilation will fail.

```
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c anet.c -o anet.o
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c mode_ac.c -o mode_ac.o
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c interactive.c -o interactive.o
interactive.c: In function ‘interactiveShowData’:
interactive.c:143:23: error: initializer-string for array of ‘char’ is too long [-Werror=unterminated-string-initialization]
  143 |     char spinner[4] = "|/-\\";
      |                       ^~~~~~~
cc1: all warnings being treated as errors
make: *** [Makefile:223: interactive.o] Error 1
make: *** Waiting for unfinished jobs....
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c dump1090.c -o dump1090.o
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c comm_b.c -o comm_b.o
x86_64-pc-linux-gnu-gcc  -I. -D_POSIX_C_SOURCE=200112L -DMODES_DUMP1090_VERSION=\"unknown\" -DMODES_DUMP1090_VARIANT=\"dump1090-fa\" -D_DEFAULT_SOURCE -DENABLE_CPUFEATURES -Icpu_features/include -DENABLE_RTLSDR -DENABLE_SOAPYSDR -DSTARCH_MIX_X86 -O2 -pipe -march=native -fno-diagnostics-color -std=c11 -fno-common -Wall -Wmissing-declarations -Werror -Wformat-signedness -W -I/usr/include/ -c mode_s.c -o mode_s.o
```

ConiKost, 2025-03-01:
I suspect, that either `char spinner[4]` should be `char spinner[5]` or `char spinner[]`.

eabase, 2026-01-29:
Still an issue!